treehouses/remote #2: Scan for devices not working on Android 6.0+

xyb994, 2017-11-20:
`11-15 18:15:08.779 W/BroadcastQueue(773): Permission Denial: receiving Intent { act=android.bluetooth.device.action.FOUND flg=0x10 (has extras) } to ProcessRecord{9fc1db4 13809:com.example.yubo.bluepi/u0a328} (pid=13809, uid=10328) requires android.permission.ACCESS_COARSE_LOCATION due to sender com.android.bluetooth (uid 1002)`

Some googling turned out with `targetSdkVersion` 23 or higher, the app needs to request `ACCESS_COARSE_LOCATION` permission from user on Android 6.0+.

Workaround is to pair with raspberry pi on the phone's Bluetooth settings, then come back and pick the already paired raspberry pi in the popup.

xyb994, 2017-11-20:
Looks like there's already an open PR addressing this issue in the sample app https://github.com/googlesamples/android-BluetoothChat/pull/4

i5o, 2017-12-28:
/me wonders if applying https://github.com/googlesamples/android-BluetoothChat/pull/4 will fix our problems.

I'll try to apply that patch in the following days :)

dogi, 2018-02-08:
https://github.com/treehouses/remote/pull/6
resolved by this or next commit

dogi, 2018-05-16:
@xyb994 you reopened this without any message

I will close it in the betweentime

xyb994, 2018-05-16:
@dogi The issue was never fixed, I saw @leomaxi commented at issue https://github.com/treehouses/remote/issues/70#issuecomment-378663938, so I referenced it here and re-opened this issue.

closed 2019-02-20 by dogi