treehouses/remote #8: Wifi - Normal [with and without password]

leomaxi, 2018-01-18:


dogi, 2018-01-18:
`pirateship wifi <ESSID> [password]`              attached to wifi hotspot
and maybe also
`pirateship staticwifi <ip> <mask> <gateway> <dns>`

interface should enable both options:
- [ ] `pirateship wifi <ESSID> <password>`
- [ ] `pirateship wifi <ESSID>`
- [ ] `pirateship staticwifi <ip> <mask> <gateway> <dns>`

Jackbui96, 2018-03-08:
I will try to work on this issue.

xyb994, 2018-03-08:
@Jackbui96 the third one actually already has an issue created, see #7

xyb994, 2018-03-08:
### User story
As a user, I would like to be able to configure the Raspberry Pi to connect to a Wi-Fi hotspot.

### Acceptance criteria
- click on the `Connect to Wi-Fi` button, a dialog appears.
- the dialog asks for the `SSID` and `Password` of the Wi-Fi network.
  - The dialog would look something similar to [this](https://camo.githubusercontent.com/8a9a8b70ce23b9ed6029ef04c7c20a145ec6e738/68747470733a2f2f7975726f636b6b6b2e6769746875622e696f2f426c7565746f6f74682d5250692d507974686f6e2f696d616765732f426c75652d5250692f776966692d636f6e66696775726174696f6e2e706e67)
  - indicate `SSID` filed is required
  - indicate `Password` filed is optional
- after the user click on `Start Configuration`, the `pirateship wifi [ESSID] <password>` should be sent to Raspberry Pi.

closed 2018-05-16 by dogi