treehouses/remote #9: Hotspot - [with and without password]

leomaxi, 2018-01-18:


dogi, 2018-01-18:
`pirateship hotspot <ESSID> [password]`              creates a mobile hotspot

interface should enable both options:
- [ ]`pirateship hotspot <ESSID> <password>`
- [ ]`pirateship hotspot <ESSID>`

xyb994, 2018-03-08:
### User story
As a user, I would like to be able to configure the Raspberry Pi in the `remote` app to run a Wi-Fi hotspot.

### Acceptance criteria
- click on the `Create W-Fi hotspot` button, a dialog appears.
- the dialog asks for the `SSID` and `Password` the user would like the hotspot to have.
  - The dialog would look something similar to [this](https://camo.githubusercontent.com/8a9a8b70ce23b9ed6029ef04c7c20a145ec6e738/68747470733a2f2f7975726f636b6b6b2e6769746875622e696f2f426c7565746f6f74682d5250692d507974686f6e2f696d616765732f426c75652d5250692f776966692d636f6e66696775726174696f6e2e706e67)
  - indicate `SSID` filed is required
  - indicate `Password` filed is optional
- after the user click on `Start Configuration`, the `pirateship hotspot [ESSID] <password>` should be sent to Raspberry Pi.

### More...
- check if hotspot is up and running?

Jackbui96, 2018-03-08:
Thank you for the User Story Yibo, that would help a lot.

commit 54c0622a (2018-03-15): create hotspot (fixes #9) (#54)
closed 2018-03-15 by dogi