treehouses/remote #14: Display of hardware version in header

lorugant, 2018-01-25:
Displaying the hardware version in the header of the app beside "Connected to raspberry pi" so that it shows like "Connected to raspberry pi 3 Model B Rev 1.2"

Going-Gone, 2018-02-19:
I would like to help out on this issue.
Could you update on where you are with this issue?
What problems are you still facing, and what have you finished so far?

kylemathias, 2018-02-20:
i can take a look. make a new branch if you want to try and add this feature.

lorugant, 2018-02-21:
Yes @Going-Gone and @kylemathias I was trying to hardcode  by sending only "pirateship detectrpi" to chatservice , create a new thread and display by creating a duplicate socket but I was getting stuck in the while loop of run method of a method which is a copied version of connectedThread. In shorter version, I was having problems with not sending but receiving from input stream of socket as it is getting either stuck in the loop or getting out of that loop with the help of state variable to UI and displaying "null". I will make a pull request so you guys will have better clarity and can contribute.

commit 5a584638 (2018-03-29): pirateship detectrpi in header (fixes #14) (fixes #22) (#58)
closed 2018-03-29 by dogi