treehouses/remote #21: Do we wanna use Material Design?

dogi, 2018-02-08:
https://developer.android.com/training/material/index.html

dogi, 2018-02-08:
Please tell me you opinion ;)

lorugant, 2018-02-08:
I think it would look really professional. It would look good

AshishJumbo, 2018-02-09:
It would help standardize stuff and make things look more consistent.

TooManySpaces, 2018-02-09:
Yes! This would be a good time.

Going-Gone, 2018-02-10:
I also think this would be a good idea to pursue.

lorugant, 2018-02-22:
[image]
[image]
[image]
[image]

Hi @/all Please check the above screenshots of templates that I made for UI design. Please review and suggest. Thanks

lorugant, 2018-02-22:
https://www.justinmind.com/ 

Hi @/all I used the baove tool for creating

xyb994, 2018-02-22:
Maybe we could consider using one of these for general navigation?
https://material.io/components/android/catalog/navigation-view/
https://material.io/components/android/catalog/bottom-navigation-view/

We need to categorize the features we would like to have in the remote app.

xyb994, 2018-02-22:
@leomaxi: We would like to collect info about who's using the app. Hidden way to skip login/sign up.

closed 2018-05-16 by dogi