treehouses/remote #22: display the software version in the header

dogi, 2018-02-08:


xyb994, 2018-02-09:
`cd /boot; cat version.txt; cd -`

Jackbui96, 2018-02-14:
Try to insert these 2 commands in BluetoothChatFragment.java. Between line 398 and 399 (according to Github), insert:

mChatService.write("cd boot\n".getBytes());
mChatService.write("cat version.txt\n".getBytes());

You should get a result like this whenever you connected your phone to the Raspberry Pi.

[image]
[image]

lorugant, 2018-02-14:
The Expected Result should be "Connected to software version release-40" is what you mean right? @Jackbui96

Jackbui96, 2018-02-14:
Yes that would be correct. But these 2 commands sometime works and sometime it doesn't, so we are still figure it out how to solve it.

Jackbui96, 2018-02-16:
So I found a way to display the version number in the heading but it kindna dirty. Please use this if you found this helpful: https://github.com/Jackbui96/Jackbui96.remote/tree/issue22-clean

Screenshot Preview: 
[image]

lorugant, 2018-02-19:
@Jackbui96  I saw the code which you wrote. If we are just using this app for testing purposes then we can use your solution but as you said, in the longer run, its going to eat up the battery. But I feel that it can be a possible solution because I tried to use the existing methods but runonuiThread is also a good way.  Also I have a suggestion, do you think we can make the line if (out.contains("release-")) to another condition? or use a looper? because then we would be able to display not only the software version but the hardware version too. What do you think?

TooManySpaces, 2018-02-21:
What if we tried something like this?

    
    private FragmentActivity mActivity;
    private String out;
    private boolean alreadyExecuted = false;

add boolean so if statement only runs once

    if(!alreadyExecuted) {
                        if (out.contains("release-")) {
                            mActivity.runOnUiThread(new Runnable() {

                                @Override
                                public void run() {
                                    final ActionBar actionBar = mActivity.getActionBar();
                                    if (null == actionBar) {
                                        return;
                                    }
                                    Log.d(TAG, "actionBar.setSubtitle(subTitle) = " + out);
                                    //currentStatus = subTitle.toString();
                                    actionBar.setSubtitle("Version: " + out);
                                }

                            });
                        }
                        //set alreadyExecuted to true so it only checks for "release-" once
                        alreadyExecuted = true;
                    }

Can't test this since I don't have a RPi, yet. 

#33 so someone can test for themselves

lorugant, 2018-02-21:
@TooManySpaces  This is already done by @Jackbui96 .Although we were able to display, it is inefficient as it will be consuming more battery power. Do you have any ideas on how we can make the thread execute it once with the help of state variable and then change the state in actionBar??

TooManySpaces, 2018-02-21:
@lorugant isn't it only running the runuiThread () once since its in a conditional loop? I put it that way so it isnt constanly running the if statement, and then constantly running the runonUiThread (). Sorry if this seems like a bad question. I'm still wrapping my head around some things.

Jackbui96, 2018-02-21:
@lorugant @TooManySpaces I am very sorry for the late reply. I was busy with my school works these couple of days, that is why I cannot reply to this thread as until now.

@TooManySpaces Thank you very much for further improving my solution. I think we can solve one of the problem. But as Lalitha said, even though it is true that you are running the execution 1 time only. The problem still remains that we are still consuming energy to check for the condition loop. So every time we get the write input, we are still consuming energy to check for the "if statement". But if we try to use Lalitha approach, which is to create a new thread that only run once and it is done. Then we don't have to worry about the energy consumption.

That is my opinion for this issue. But for the moment, we can use your approach. But for optimization and performance enhancement, we will need to switch to Lalitha's approach in the future.

TooManySpaces, 2018-02-21:
@Jackbui96 Ah, okay, thanks for clearing it up for me.

xyb994, 2018-03-01:
[image]
[image]

xyb994, 2018-03-05:
@Jackbui96 detectrpi branch worked well on the 10 inch OLE tablet.

https://user-images.githubusercontent.com/7483137/36995599-83f2fcaa-2082-11e8-98b7-8fa6d7bf9600.png
https://user-images.githubusercontent.com/7483137/36995600-8404ef50-2082-11e8-9d87-47eaedb7472f.png

Jackbui96, 2018-03-05:
@xyb994 Thank you for testing Yibo, those are very good results. Right now me and Daren are trying to make sure that the 3 commands will always work.

commit 5a584638 (2018-03-29): pirateship detectrpi in header (fixes #14) (fixes #22) (#58)
closed 2018-03-29 by dogi