treehouses/remote #23: Running Android app in Virtual Device

lorugant, 2018-02-08:
The app gets crashed when trying to run on a Virtual Device. 

02-08 17:59:11.263 2383-2383/? E/AndroidRuntime: FATAL EXCEPTION: main
                                                 Process: io.treehouses.remote, PID: 2383
                                                 java.lang.RuntimeException: Unable to start activity ComponentInfo{io.treehouses.remote/io.treehouses.remote.MainActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.bluetooth.BluetoothAdapter.isEnabled()' on a null object reference
                                                     at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2416)
                                                     at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2476)
                                                     at android.app.ActivityThread.-wrap11(ActivityThread.java)
                                                     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1344)
                                                     at android.os.Handler.dispatchMessage(Handler.java:102)
                                                     at android.os.Looper.loop(Looper.java:148)
                                                     at android.app.ActivityThread.main(ActivityThread.java:5417)
                                                     at java.lang.reflect.Method.invoke(Native Method)
                                                     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
                                                     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
                                                  Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.bluetooth.BluetoothAdapter.isEnabled()' on a null object reference
                                                     at io.treehouses.remote.BluetoothChatFragment.onStart(BluetoothChatFragment.java:136)
                                                     at android.support.v4.app.Fragment.performStart(Fragment.java:2369)
                                                     at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1458)
                                                     at android.support.v4.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManager.java:1740)
                                                     at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1809)
                                                     at android.support.v4.app.FragmentManagerImpl.dispatchStateChange(FragmentManager.java:3217)
                                                     at android.support.v4.app.FragmentManagerImpl.dispatchStart(FragmentManager.java:3176)
                                                     at android.support.v4.app.FragmentController.dispatchStart(FragmentController.java:203)
                                                     at android.support.v4.app.FragmentActivity.onStart(FragmentActivity.java:562)
                                                     at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1237)
                                                     at android.app.Activity.performStart(Activity.java:6253)
                                                     at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2379)
                                                     at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2476) 
                                                     at android.app.ActivityThread.-wrap11(ActivityThread.java) 
                                                     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1344) 
                                                     at android.os.Handler.dispatchMessage(Handler.java:102) 
                                                     at android.os.Looper.loop(Looper.java:148) 
                                                     at android.app.ActivityThread.main(ActivityThread.java:5417) 
                                                     at java.lang.reflect.Method.invoke(Native Method) 
                                                     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726) 
                                                     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)

dogi, 2018-02-08:
@lorugant thanks - I also assume that it crashes because there is no Bluetooth in the Virtual Device

TooManySpaces, 2018-02-09:
Yeah, the emulator doesn't have Bluetooth, and I think the program is trying to connect through bluetooth on startup, so that's why it crashes when it starts.