treehouses/remote #31: Fix the activity device list layout

Going-Gone, 2018-02-19:
This may have to do with specific device that I am using, but the "select device to connect" is partially cut off and could be made to look more neat.  

[image]

leomaxi, 2018-02-19:
Good catch @Going-Gone. Can you include the device type / detailed info?. That will be easier for us to troubleshoot / fix it.

Going-Gone, 2018-02-26:
https://stackoverflow.com/questions/2711858/is-it-possible-to-set-a-custom-font-for-entire-of-application/16883281#16883281 If anyone has some input on this, any suggestions would be appreciated. I am not sure if this is possible or not. This link describes a bit on this, but I am not sure if we can incorporate it into remote.

Going-Gone, 2018-03-02:
The area that is clipping, is actually the ActionBar, and it is connected to the Theme.Holo.Dialog within the AndroidManifest.xml. 

What we could do is, use Theme.Holo.Dialog.NoActionBar, and then create a custom progress bar, and call it when needed.

kylemathias, 2018-03-12:
I think we are in the process of choosing a different format to start our app over. I would advise to use something more recent than Android 4.