treehouses/remote #43: rename hostname

xyb994, 2018-03-08:
```
pirateship rename <hostname>                       changes hostname
```

Going-Gone, 2018-03-08:
Do you want this command to be inside a button? If so does it matter where the button is at? (action bar, or with the other command buttons)

xyb994, 2018-03-08:
@Going-Gone Since Lalitha is working on the UI/UX now, put it in a button with the other command buttons for now

xyb994, 2018-03-08:
### User story
As a user, I would like to be able to change the hostname of the Raspberry Pi in the `remote` app

### Acceptance criteria
- click on the `Rename hostname` button, a dialog appears.
- the dialog asks for the new `hostname`
  - The dialog would look something similar to [this](https://camo.githubusercontent.com/8a9a8b70ce23b9ed6029ef04c7c20a145ec6e738/68747470733a2f2f7975726f636b6b6b2e6769746875622e696f2f426c7565746f6f74682d5250692d507974686f6e2f696d616765732f426c75652d5250692f776966692d636f6e66696775726174696f6e2e706e67)
  - indicate `SSID` filed is required
  - in the lower right corner of dialog, `Change hostname` should be grayed out if there's nothing in the `SSID` filed
- after the user click on `Change hostname`, the command `treehouses rename <hostname>` should be sent to Raspberry Pi.

### More...
- remind user to reboot for it to take effect?
- remind user that bluetooth name would change too?

commit 445994e0 (2018-03-15): change hostname (fixes #43) (#47)
closed 2018-03-15 by dogi