treehouses/remote #51: pirateship wifi <SSID> <password> not working

Jackbui96, 2018-03-13:
I don't know how the script work. But when I try to send this string "pirateship wifi nameHere passwordHere", it said that the command is not supported. On the other hand, this command will work "{"SSID":"namehere","PWD":"passwordhere"}". Does any vi here have any idea? Please feel free to share your opinion.

leomaxi, 2018-03-15:
I had a similar situation with release 41, using the bluetooth app. But the command worked fine for me when I did it directly through ssh.

xyb994, 2018-03-15:
"command is not supported" is most likely due to the regex validation [here](https://github.com/ole-vi/bluetooth-server/blob/master/bluetooth-server.py#L41)

@Jackbui96 similar to the issue you ran into yesterday during the hangout:
[image]

closed 2018-05-16 by leomaxi