treehouses/remote #70: Bluetooth Connection fail

kylemathias, 2018-03-26:
I am getting this error box when I try to connect my android galaxy s7 device to my RBpi 3.
[image]

leomaxi, 2018-04-04:
[image]
[image]

`Bluetooth devices doesn't show up on Pixel 2 (Android 8) but shows up on Infinix (Android 5.1)  but when go through android's bluetooth menu, all bluetooth devices show up`

xyb994, 2018-04-04:
@leomaxi The problem you encountered seem to be related to issue #2

closed 2018-05-16 by dogi