treehouses/remote #73: Create a "disconnect" button for the user.

Jackbui96, 2018-03-28:
I noticed that we don't have a disconnect button for the user to use. Because at some point the user will want to disconnect the pi. With that being said, what do other vi think of this implementation? Do we really need to have a disconnect button?

kylemathias, 2018-04-02:
We could have a disconnect button (i don't see why not). we could add it to the connection button. after we are connected, if the user clicks on connection button again they could then be giving a "disconnect" button.

Jackbui96, 2018-04-02:
@kylemathias Thank you for your feedback Kyle, then I think our "connect a device - insecure" would serves that purpose. For me, if I try to "connect" to a device when I am already connected. Then I will get disconnected. So I think we can try to change the String into something like "connect/disconnect a device - insecure".