treehouses/remote #90: Steps to get Bluetooth connected to RBi3

kylemathias, 2018-04-12:
I recently started the process of getting myself connected to the RBi3. From my adventures, I have done a lot of different things to try and get our remote app working with the RBi, and I thought I should document the things that need to be done to get it working. We should try and finds ways to avoid this process to simplify the connection steps.

Img: treehouse-44.img
Android Device: Galaxy s7
Pi: RBi3

My Bluetooth connection was not stable. I would be able to pair my device with the RBi but it would never stay connected.

Is there something i am missing? what steps have you used to get your Android device connected and stable to treehouse .img RBi?

kylemathias, 2018-04-16:
When I say stable I mean I see the RBPi Bluetooth status constantly turn from red to green about every other second.

dogi, 2018-05-16:
@kylemathias please test new image -> it has a new bluetooth server

closed 2018-05-16 by dogi