treehouses/remote #91: Connect to Wi-Fi hotspot not working

xyb994, 2018-04-13:
### Problem
Not able to connect to Wi-Fi hotspot using the Wi-Fi button on the upper right corner.
- The command gets send is still including `pirateship hotspot`.

### Screenshot
[image]

Going-Gone, 2018-04-16:
It seems that we have two issues here with our latest release. 

1. The Wifi configuration and Hotspot configuration commands are being sent together. This issue popped up after release version (v124) after the hotspot dialog was created. 

2. In releases prior to the hotspot functionality, the Wifi configuration disconnects the PI device while configuring the Wifi.

Going-Gone, 2018-04-19:
[image]
Could this be the reason for the double message being sent?

commit 31596f84 (2018-04-26): separation of Hotspot and Wifi form (fixes #91) (#97)