treehouses/remote #113: features to consider for design

xyb994, 2018-06-07:
Connect to a device

#### Network
  - Ethernet: `<ip> <mask> <gateway> <dns>`
  - Wi-Fi: `<ESSID> [password]`
    - Static Wi-Fi: `<ip> <mask> <gateway> <dns> <ESSID> [password]`
  - hotspot: `<ESSID> [password]`
  - bridge: `<ESSID> <hotspotESSID> <password> [hotspotPassword]`
  - Reset: confirm will reset following: hostname, turn bluetooth off, country setting, all network related change

#### Docker
  - display docker containers: a toggle for showing all/running containers
```
CONTAINER ID        IMAGE               COMMAND                  CREATED             STATUS                      PORTS               NAMES
e3cbdea7a4ad        housing_django      "/entrypoint.sh ./co…"   7 months ago        Exited (137) 7 months ago                       housing_django_1
1172328636be        housing_postgres    "docker-entrypoint.s…"   7 months ago        Exited (0) 7 months ago                         housing_postgres_1
```
  - display all docker images?
```
REPOSITORY                     TAG                                      IMAGE ID            CREATED             SIZE
treehouses/planet-test         latest                                   ac7ec36b158b        5 weeks ago         20.6MB
treehouses/planet-test         db-init                                  93c9db361107        5 weeks ago         79MB
treehouses/planet              0.1.8                                    85f02e4349c0        5 weeks ago         20.6MB
treehouses/planet              db-init-0.1.8                            70ebba4044ee        5 weeks ago         79MB
treehouses/couchdb             1.7.1                                    ab618d800983        2 months ago        223MB
treehouses/couchdb             2.1.1                                    612df5f25616        2 months ago        333MB
<none>                         <none>                                   350c188acf3d        3 months ago        722MB
result/latest                  latest                                   e4322f21e6c2        3 months ago        840MB
treehouses/moodle              rpi-alpine-0.0.2-better_nginx-1bba7c8e   dc1339b76748        3 months ago        252MB
treehouses/moodle              alpine-0.0.2-better_nginx-1bba7c8e       a86ff28df715        3 months ago        255MB
```

#### System
  - reboot: confirm
  - expand file system: confirm and remind to reboot
  - rename hostname: user input new hostname
  - change password for 'pi' and 'root' user: user input new password
  - container non | docker | balena: choose 1 of 3 options
  - upgrade cli

#### System Configuration
  - set timezone 
  - set locale 
  - ssh enable/disable
  - vnc enable/disable

#### Terminal
Functions like a normal terminal, a warning dialog before proceed?

#### About (look for a library for this?) (not priority at this time)
  - app intro/help
  - license
  - contributor

---

This list left out sshkeyadd and bluetooth from cli.

aakarnik, 2018-07-02:
All the images are uploaded and stored in this [folder](https://drive.google.com/drive/folders/1xWspOq_ferSQuls6YMfIXCb4lLIEeANP?usp=sharing)

xyb994, 2018-07-03:
InVision preview of the new design looks nice, I made some comments for consideration.

aakarnik, 2018-07-11:
Update: [New Link](https://invis.io/JEMWGN7KQ68#/308172028_TRDesign-Draft3-1)

aakarnik, 2018-07-27:
Update: [New Link](https://invis.io/86N80OBZRF5#/311325273_TRDesign-Draft4-12)

aakarnik, 2018-08-06:
Update: [Final Draft](https://invis.io/WAMBXA8BFRD#/312866152_TRDesign-Draft5-1)

closed 2020-02-27 by dogi