treehouses/remote #146: Please increase versionName/versionCode

IzzySoft, 2019-01-11:
All releases identify as 2.0/2 – and hence an update is never detected. Your last tag is 528, so I assume that should be the current versionCode. Could you please increase at least that (corresponding to the tag name) with each new release/tag? Thanks in advance!

dogi, 2019-02-20:
@IzzySoft just saw this issue ... sorry

anyhow wonder if https://github.com/treehouses/remote/commit/786bda87e44942188059a0fe7f08168e59772aaf fixed this issue?

IzzySoft, 2019-02-21:
I doubt that changing a file name will do the trick (unless it means that until now it was always the very same APK attached to all tags, and now it will be the correct one). The problem [lies here](https://github.com/treehouses/remote/blob/master/app/build.gradle#L9) – in this place you need to update `versionCode` and `versionName`. You see, there is the 2.0 (versionName) and 2 (versionCode). I initially mentioned.

IzzySoft, 2019-02-23:
@dogi any chance? At least for the "green releases" (if not for the pre-releases as well)? Just use the same number as you use for the tag name (without the leasing "v").

Sriharsha-Singam, 2019-02-23:
I think we can make a pull request before releasing to F-Droid by manually change the Version Code and Version name in the Gradle File, or if it is possible we can then link the Travis Version number to the VersionCode in the gradle file.

@dogi @empeje @IzzySoft

IzzySoft, 2019-02-23:
Oh, releasing to F-Droid would of course be much appreciated (I see no corresponding [RFP](https://gitlab.com/fdroid/rfp/issues) yet – but of course it's fine to open that once you're ready). Currently, [Treehouses Remote is in my repo](https://apt.izzysoft.de/fdroid/index/apk/io.treehouses.remote) – and I was asking for 2 reasons: without the versions being incremented, 1) I have to manually replace the file at each update as otherwise fdroidserver refuses to accept it as it looks to be the very same version, and 2) users never get an update notification (as for Android, two APKs with the same versionCode are the same version).

dogi, 2019-02-28:
https://proandroiddev.com/keep-salinity-with-your-versioncode-db2089b640b9

dogi, 2019-02-28:
We should also consider change of https://github.com/treehouses/remote/blob/master/.travis.yml#L53 based on versionname

IzzySoft, 2019-02-28:
@dogi I've seen this kind of calculation in several repos already. I like the idea. Unfortunately, if you put the formula into your `build.gradle` and let gradlew do the math, F-Droid won't be able to auto-update (the updater just *parses* the `build.gradle`, it doesn't *run* it). If however you do the math yourself and put the result in *literally* (like `versionCode: 1003000`), that would work fine indeed.

IzzySoft, 2019-05-18:
Just wondering: is it really that hard to at least increment versionCode? The new release still has the very same issue, and thus will not be seen as update if the previous version was installed on a device.

IzzySoft, 2019-07-01:
I see your new CI releases solved this. Though now `versionName` looks a bit "unusual", but that's fine with me :smile: Shall we close this issue then?

dogi, 2019-07-01:
@IzzySoft thanks

closed 2019-07-01 by dogi