treehouses/remote #165: detect in terminal certain commands which never stop

dogi, 2019-01-28:
- top
- iotop
- ping
- screen
- vim
- less
- ...

and give feedback
in form of this command are blocked in this serial line by line interface cause they never terminate

other question is should we do this in https://github.com/treehouses/remote
or https://github.com/treehouses/control

i5o, 2019-01-30:
we can do this in remote to avoid sending stuff to the server, or we can do this in control and __control__ this on the code, which would make things slower (one more useless request).

we could also think to find a way to detect 'interactive' commands on control and avoid them at all

Sriharsha-Singam, 2019-02-11:
Is it possible to have a more comprehensive list of interactive terminal commands or is that it? @dogi @i5o

i5o, 2019-02-13:
There is no specific way to know if a command requires user input (or maybe I'm wrong, correct me if so), but these were just some examples we provided.

The "idea" was to timeout them; perhaps a limit of 10 seconds is enough?

i5o, 2019-02-13:
also perhaps disable the limit for "treehouses *"?

i5o, 2019-02-13:
this needs to be done in @treehouses/control I think

SeanTang001, 2020-09-20:
This issue is pretty bad. It seems and htop and top had been fixed for now. However, ping still results in a never ending stream. Aside from breaking "terminal", it also breaks other functions such as "status".