treehouses/remote #166: Upload to playstore

empeje, 2019-01-28:
How about we upload the app to play store?

IzzySoft, 2019-02-14:
I'd like it more if it would be available at [F-Droid](https://f-droid.org/), which is THE place for Android FOSS apps (and other than Playstore, it's privacy friendly). I took the freedom to add it [to my repo](https://apt.izzysoft.de/fdroid/index/apk/io.treehouses.remote) for now. To get it into the main repo requires your consent – and the project to specify its license. Feel free to ping me if you need help with the former 😃

empeje, 2019-02-18:
Thank you @IzzySoft for the suggestion, I like the idea about the F-Droid.

IzzySoft, 2019-02-18:
@empeje Great! Just let me know when you decided on a license, and want me to get the process started. Disclosure: I'm one of the maintainers there :smile:

closed 2019-06-30 by inDepthh

IzzySoft, 2019-07-01:
So with the issue closed now, are you still aiming for the official F-Droid repo? And seeing you changed the way `releases/` look (no longer "2.0,2" but rather a real versionCode which is versionName at the same time), with daily releases, are they all considered "releases" or rather "nightly builds" – and in the latter case, how to tell which is a "real release" and which I should pick as update in my repo (see above)?

inDepthh, 2019-07-01:
@IzzySoft Every PR it releases automatically. PRs are typically merged every Monday, Wednesday, and Friday, so 2 - 3 releases a week (not including consecutive releases in one day). im not sure how often it is updated in google play, but i think its every release. We also might want to consider changing the way we do this @dogi

IzzySoft, 2019-07-01:
Thanks for the fast response, @inDepthh – I guess that means I should enable auto-updates then (now that `versionCode` is increased with each update, I can do so) and fetch each one?

inDepthh, 2019-07-01:
@IzzySoft  @dogi It feels like we would be spamming users with updates, this is why i suggest we change the process to one release a week. The argument can also be made that we really dont have many users right now.. so maybe once we release the app from beta we should do releases in a more traditional way.. you know, what we have now is good for testing and allows for faster feedback.

inDepthh, 2019-07-01:
in other words, for now its ok to do auto updates i think. and when/if we change the release timeline, it should not change the auto update part

IzzySoft, 2019-07-01:
I've enabled auto updates for now. If you have the feeling this "spams" users, reducing releases to once per week is one way; another option would be declaring the "nightly builds" to be "pre-releases" (which is an option with Github releases) and manually setting "stable" ones as releases (i.e. "untick" the "pre-release" option). I can tell my updater to ignore pre-releases, which is the default anyway. That's what is configured for your app here already, so it would work automatically if you set this up on your end.

While on it: I've noticed there are 2 APKs attached, one of them seems to be unsigned. So I need to make sure to only grab the other. To assure that, I've told my updater to only grab APKs whose file names match the RegEx `\d+\.apk$` (like `remote-1200.apk`), which currently is the signed one. Should you plan to change the naming schema, please let me know.

IzzySoft, 2019-07-04:
Updater just failed as this time the signed version was "the other". So I switched that in my config as well. Can you please make sure you always sign the same one (currently `remote.apk`)? Hard to automate updates otherwise, @inDepthh :wink: