treehouses/remote #177: Often times the Bluetooth connection dialog does not dismiss on success

inDepthh, 2019-02-02:
Sometimes is does dismiss, but sometimes it does not
[image]

This is shows the dialog continuing after connection

Sriharsha-Singam, 2019-02-02:
How to reproduce?

inDepthh, 2019-02-02:
It happens on phyical tablet. Im not sure why it works sometimes and not others. After regular testing with connecting to Bluetooth you will likely run into it. I’m going to look into to see if I can give more information. I’m guessing that it could be caused by a condition that does not always get evaluated to true, therefore not calling dismiss, but that’s a guess.

PR #260 (2019-05-09): stabilize Bluetooth connection (fixes #177)
fixes #177
commit 9899b9d0 (2019-05-09): stabilize Bluetooth connection (fixes #177) (#260)
closed 2019-05-09 by dogi