treehouses/remote #199: Show recent commands

rrijal53, 2019-02-27:
I send some commands and waited for some time, then my screen goes off.
 When i turned on my screen i'm not able to see recent commands and their output.

[image]

Sriharsha-Singam, 2019-02-27:
I guess we could keep all commands in the listview no matter what and then when you want to get rid of everything -- we could add a clear button to clear out all the past commands

rrijal53, 2019-02-27:
@Sriharsha-Singam  yes, that is the easiest solution.

inDepthh, 2019-03-11:
@rrijal53 Should we use realm database to save the commands? If we use the database, we can then use arrow keys (up or down) to cycle through recent commands like real terminals do, as well as save them in the list view until it is cleared

commit 21df671b (2019-04-04): terminal: save recent commands (fixes #199) (#221)
closed 2019-04-04 by dogi