treehouses/remote #388: tor doesn't configure properly with Raspberry Pi Zero W

Kid243, 2019-07-15:
These command would display success operations:
treehouses tor start
treehouses tor add 

However when using these commands:
treehouses tor list
treehouses tor notice on

The following error ensues:

Error: the tor service has not been configured.
Run 'treehouses tor start' to configure it.
Or 'treehouses add [localPort]' to add a port and be able to use the service

[image]

dogi, 2019-07-15:
[image]

sc4773rm1nd, 2019-07-15:
Yeah I get it too
[image]

sc4773rm1nd, 2019-07-15:
Same in terminal
[image]

dogi, 2019-07-21:
https://github.com/treehouses/builder/pull/475

closed 2019-07-21 by dogi