treehouses/remote #469: where should treehouses rename go?

dogi, 2019-09-07:
[image]

ChiragAwale, 2019-09-08:
Command list different category maybe?

dogi, 2019-09-08:
no changing hostname of rpi is something more intense then just a command in the list of prewritten ones

ChiragAwale, 2019-09-16:
@dogi I am sorry I do not get it, is it not a one line command like others though?

dogi, 2019-09-17:
@ChiragAwale yes and no ... most actions of `treehouses remote` are cli or combination of cli actions - I wonder mostly under which menu section we should put `treehouses rename <newhostname>`

rrijal53, 2019-10-17:
@dogi  @ChiragAwale  We will introduce new button on status page somewhere here ..
[image]

commit b46ca054 (2019-11-21): status: add rename (fixes #469) (#505)
closed 2019-11-21 by dogi