treehouses/remote #480: Release 1598 adds proprietary components and tracking

IzzySoft, 2019-09-25:
Could it be this slipped in accidentally? Suddenly, the app comes with GMS, Firebase and Firebase Analytics. Would be great if at least one of the 2 APKs attached to `releases/` could come without that. Is there a chance?

IzzySoft, 2019-09-28:
Thanks for replacing the APK file – but the replacement seems to lack v1 signatures. `apksigner verify` reports:

```
DOES NOT VERIFY
ERROR: No JAR signatures
```

Which means v1 (JAR) signatures are missing. Without those, the minVer of Android 4.0.3 is invalidated: v2 signatures were introduced with Android 7 (or 8?), so older versions cannot deal with those.

Could you please fix that up? Thanks!

dogi, 2019-10-02:
> Could it be this slipped in accidentally? Suddenly, the app comes with GMS, Firebase and Firebase Analytics. Would be great if at least one of the 2 APKs attached to `releases/` could come without that. Is there a chance?

We definitely share your concerns regarding the privacy of users, but given the limited resources of our organization tend to use the most cost efficient solutions which at present seems to be google analytics, we would however love to hear any ideas you may have to allow more user control over their data, is this something you could assist with?

dogi, 2019-10-02:
@IzzySoft we reverted it ... it was an experiment and have to say it had a fast turnaround. Anyhow wonder what you are thinking about https://github.com/treehouses/remote/pull/476

IzzySoft, 2019-10-02:
Thanks, @dogi!

> we would however love to hear any ideas you may have to allow more user control over their data

If you need a privacy friendly analytics framework, at F-Droid we recommend [ACRA](https://github.com/ACRA/acra). Google lives off data, and their stuff is proprietary. You cannot guarantee the users' privacy, even with best efforts from your end. You can never tell for sure a) what data will be collected and b) what will happen with it. With ACRA you can, as it's FOSS. And comes with privacy-friendly defaults.

> we reverted it

That's a welcome turn! :green_heart: My updater just fetched 1612 a few hours ago, and indeed the trackers are gone (and so is the signing error). Yay! :tada:

> Anyhow wonder what you are thinking about #476

I'm no Android dev so I cannot review that PR, sorry. The technical implementation apart: this should be opt-in, i.e. the collection should only happen *after* the user agreed (e.g. having a boolean store the decision: 0=don't send, 1=send, `null` = ask user first; that way, when `null` you can ask permission by pop-up and not have to repeat the pop-up. But give the option in settings to re-decide). With that guaranteed (and the user being aware of what exactly is collected), plus the library being FOSS… oh wait, "All rights reserved" isn't FOSS. So we're back at where we started. But it's BSD? I'm confused…

closed 2020-02-27 by dogi