treehouses/remote #521: Incorrect network report

phamduchongan93, 2019-11-28:
`treehouses internet` output `true` as the result, but there isn't any internet provided.

sriharivishnu, 2019-11-30:
@phamduchongan93 could you please clarify? Under what conditions did this occur? What did `treehouses networkmode` return?

phamduchongan93, 2019-11-30:
@sriharivishnu I have to find a way to trigger the bug again. Right now, i'm having networking issue.

sriharivishnu, 2019-12-05:
There may be something wrong with the order of input coming in sometimes: 

```
2019-12-04 21:23:26.748 12988-12988/io.treehouses.remote D/StatusFragment: writeMessage = treehouses internet
2019-12-04 21:23:26.817 12988-13083/io.treehouses.remote D/BluetoothChatService: out = 2964940 size of out = 8, bytes = 8
2019-12-04 21:23:26.818 12988-12988/io.treehouses.remote E/StatusFragment: READ
2019-12-04 21:23:26.819 12988-12988/io.treehouses.remote D/StatusFragment: readMessage = 2964940 
2019-12-04 21:23:26.823 12988-12988/io.treehouses.remote E/StatusFragment: **2964940 **false
2019-12-04 21:23:27.227 12988-13083/io.treehouses.remote D/BluetoothChatService: out = true size of out = 5, bytes = 5
2019-12-04 21:23:27.228 12988-12988/io.treehouses.remote E/StatusFragment: READ
2019-12-04 21:23:27.229 12988-12988/io.treehouses.remote D/StatusFragment: readMessage = true 
```
The memory info is coming in place of the network status

closed 2020-02-27 by dogi