treehouses/remote #710: services: no logo shows up in detail view

dogi, 2020-04-11:
[image]

dogi, 2020-04-11:
Alps TK-E102
image: 127
cli: 1.17.28
remote: 2346

sriharivishnu, 2020-04-23:
```
2020-04-22 22:08:46.216 30509-30509/io.treehouses.remote W/System.err: org.xmlpull.v1.XmlPullParserException: name expected (position:START_TAG <pathfill>@1:4147 in java.io.InputStreamReader@4486a86) 
2020-04-22 22:08:46.216 30509-30509/io.treehouses.remote W/System.err:     at com.caverock.androidsvg.SVGParser.parseUsingXmlPullParser(SVGParser.java:761)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at com.caverock.androidsvg.SVGParser.parse(SVGParser.java:603)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at com.caverock.androidsvg.SVG.getFromString(SVG.java:165)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at io.treehouses.remote.Fragments.ServiceCardFragment.showIcon(ServiceCardFragment.java:123)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at io.treehouses.remote.Fragments.ServiceCardFragment.onCreateView(ServiceCardFragment.java:60)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at androidx.fragment.app.Fragment.performCreateView(Fragment.java:2600)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:881)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at androidx.fragment.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManagerImpl.java:1238)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at androidx.fragment.app.BackStackRecord.executeOps(BackStackRecord.java:434)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at androidx.fragment.app.FragmentManagerImpl.executeOps(FragmentManagerImpl.java:2079)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at androidx.fragment.app.FragmentManagerImpl.executeOpsTogether(FragmentManagerImpl.java:1869)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at androidx.fragment.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManagerImpl.java:1824)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at androidx.fragment.app.FragmentManagerImpl.execSingleAction(FragmentManagerImpl.java:1696)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at androidx.fragment.app.BackStackRecord.commitNowAllowingStateLoss(BackStackRecord.java:299)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at androidx.fragment.app.FragmentStatePagerAdapter.finishUpdate(FragmentStatePagerAdapter.java:259)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at androidx.viewpager.widget.ViewPager.populate(ViewPager.java:1244)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at androidx.viewpager.widget.ViewPager.populate(ViewPager.java:1092)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at androidx.viewpager.widget.ViewPager$3.run(ViewPager.java:273)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at androidx.viewpager.widget.ViewPager.completeScroll(ViewPager.java:2005)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at androidx.viewpager.widget.ViewPager.onInterceptTouchEvent(ViewPager.java:2135)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at io.treehouses.remote.Views.ServiceViewPager.onInterceptTouchEvent(ServiceViewPager.java:31)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2609)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3060)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2698)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3060)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2698)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3060)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2698)
2020-04-22 22:08:46.217 30509-30509/io.treehouses.remote W/System.err:     at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3060)
2020-04-22 22:08:46.218 30509-30509/io.treehouses.remote W/System.err:     at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2698)
2020-04-22 22:08:46.218 30509-30509/io.treehouses.remote W/System.err:     at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3060)
2020-04-22 22:08:46.218 30509-30509/io.treehouses.remote W/System.err:     at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2698)
2020-04-22 22:08:46.218 30509-30509/io.treehouses.remote W/System.err:     at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3060)
2020-04-22 22:08:46.218 30509-30509/io.treehouses.remote W/System.err:     at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2698)
[... cut ...]
2020-04-22 22:08:46.219 30509-30509/io.treehouses.remote W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:7356)
2020-04-22 22:08:46.219 30509-30509/io.treehouses.remote W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
2020-04-22 22:08:46.219 30509-30509/io.treehouses.remote W/System.err:     at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
2020-04-22 22:08:46.219 30509-30509/io.treehouses.remote W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:930)
```

closed 2021-05-22 by rrijal53