treehouses/remote #788: 2020-05-25 10:45:44.955 14958-14958/io.treehouses.remote E/AndroidRuntime: FATAL EXCEPTION: main

dogi, 2020-05-27:
2020-05-25 10:45:44.955 14958-14958/io.treehouses.remote E/AndroidRuntime: FATAL EXCEPTION: main
    Process: io.treehouses.remote, PID: 14958
    android.view.InflateException: Binary XML file line #68: Binary XML file line #68: Error inflating class Button
    Caused by: android.view.InflateException: Binary XML file line #68: Error inflating class Button
    Caused by: android.content.res.Resources$NotFoundException: Drawable io.treehouses.remote:color/start_config with resource ID #0x7f060207
    Caused by: android.content.res.Resources$NotFoundException: File res/color/start_config.xml from drawable resource ID #0x7f060207
        at android.content.res.ResourcesImpl.loadDrawableForCookie(ResourcesImpl.java:898)
        at android.content.res.ResourcesImpl.loadDrawable(ResourcesImpl.java:677)
        at android.content.res.Resources.loadDrawable(Resources.java:912)
        at android.content.res.TypedArray.getDrawableForDensity(TypedArray.java:955)
        at android.content.res.TypedArray.getDrawable(TypedArray.java:930)
        at android.view.View.<init>(View.java:5442)
        at android.widget.TextView.<init>(TextView.java:1118)
        at android.widget.Button.<init>(Button.java:181)
        at android.widget.Button.<init>(Button.java:156)
        at androidx.appcompat.widget.f.<init>(AppCompatButton.java:72)
        at androidx.appcompat.widget.f.<init>(AppCompatButton.java:68)
        at androidx.appcompat.app.AppCompatViewInflater.createButton(AppCompatViewInflater.java:192)
        at androidx.appcompat.app.AppCompatViewInflater.createView(AppCompatViewInflater.java:111)
        at androidx.appcompat.app.h.a(AppCompatDelegateImpl.java:1407)
        at androidx.appcompat.app.h.onCreateView(AppCompatDelegateImpl.java:1457)
        at android.view.LayoutInflater$FactoryMerger.onCreateView(LayoutInflater.java:198)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:811)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:769)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:902)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:863)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:554)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:461)
        at io.treehouses.remote.g.q.a(DialogWifiBinding.java:73)
        at io.treehouses.remote.a.z.k.m.onCreateView(WifiBottomSheet.java:32)
        at androidx.fragment.app.Fragment.performCreateView(Fragment.java:2698)
        at androidx.fragment.app.r.a(FragmentStateManager.java:320)
        at androidx.fragment.app.m.a(FragmentManager.java:1187)
        at androidx.fragment.app.m.j(FragmentManager.java:1356)
        at androidx.fragment.app.m.i(FragmentManager.java:1434)
        at androidx.fragment.app.m.a(FragmentManager.java:1497)
        at androidx.fragment.app.a.f(BackStackRecord.java:447)
        at androidx.fragment.app.m.a(FragmentManager.java:2169)
        at androidx.fragment.app.m.b(FragmentManager.java:1992)
        at androidx.fragment.app.m.c(FragmentManager.java:1947)
        at androidx.fragment.app.m.c(FragmentManager.java:1849)
        at androidx.fragment.app.m$d.run(FragmentManager.java:413)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7100)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:964)
     Caused by: org.xmlpull.v1.XmlPullParserException: Binary XML file line #3: <item> tag requires a 'drawable' attribute or child tag defining a drawable
        at android.graphics.drawable.StateListDrawable.inflateChildElements(StateListDrawable.java:190)
        at android.graphics.drawable.StateListDrawable.inflate(StateListDrawable.java:122)
        at android.graphics.drawable.DrawableInflater.inflateFromXmlForDensity(DrawableInflater.java:146)
        at android.graphics.drawable.Drawable.createFromXmlInnerForDensity(Drawable.java:1359)
        at android.graphics.drawable.Drawable.createFromXmlForDensity(Drawable.java:1318)
        at android.content.res.ResourcesImpl.loadDrawableForCookie(ResourcesImpl.java:879)
2020-05-25 10:45:44.955 14958-14958/io.treehouses.remote E/AndroidRuntime: 	... 42 more

_Originally posted by @jiaxuan-2 in https://github.com/treehouses/remote/pull/785#issuecomment-633349730_

sriharivishnu, 2020-05-27:
@dogi could your reproduce this error?

sriharivishnu, 2020-05-28:
Try `File>Invalidate Caches and Restart`

PR #789 (2020-05-29): network: no wifi/bridge crash (fixes #788)
# fixes #788
commit 75e85127 (2020-05-29): network: no wifi/bridge crash (fixes #788) (#789)
closed 2020-05-29 by dogi