treehouses/remote #799: strange error

law-lin, 2020-06-01:
Got this error after tapping between Services and Network.
Was on the `centralize-bluetooth-3` branch
[image]

law-lin, 2020-06-03:
It's happening on the `master` branch as well. 
[image]

law-lin, 2020-06-03:
Here's another view of the error, with another error that seems to be under it.

[image]

[image]

Here's what was logged to Android Studio:
```
E/BOOLEAN: true
D/BluetoothChatService: out = ce+yxGD58uC35TzqdRjKZ5AmIrDkkentqsuDQCYfDCAQCeT0illOktLQURIRYLMZ7XUDWut26dSv+8z//E59//jnWrVvHe2K5WOtyapoGp9PJE0ex19g6NU3DxIkT8b3vfQ9XXnklWlpauJVPRNB1nec313UdTqdTWNg9iBDsTtIdl8iaNWvwzDPPYOHChfjyyy9tLg6rEDP3RW5+D3azyrKMsrIyjB8/Ht/4xjdwxRVXwDRNLsaGYSASifCMfB1BRLbvCvo3rHCBLMu2B3Mmk+EDkqZp8kIVsizzh/OuXbvwz3/+E88//zxWrVoF0zShaRrS6TR0XYeqqlxgAdiSibH/VVXlRSsAIBgMoqysDPfddx9Gjx6Nk08+GaqqQtd1PkjKtkUIds8hBLsTKLICM22iJFCCcFsYtTW12LdvH5b/618YNWoUZFVBOp2GaZpwuVxwOBwIh8M8LG/Tpk3YuHFjwcraqqry0CuXy4V0Os3/DwaDiMViSKVSqK2txRVXXIGZM2di7NixALLZ35jIt4fwVR/dtLW18Yx80WgU//znP/G3v/0Ny5cvR3NzM/dtA7AlnwKy1yMTbSvMjy5JEgzDwIgRI3DppZdi6tSpGDduHDcWkskkEokEgsEgAGDnzp152R27c/8eLfpjRQj2wdbvkJHOZAfrfF4forEoJEggED5Z/SlGnnoq0pkD5aR0XceyZcswZ84cvPLKKwgEAmhra+Pr8/l83OpIJBLwer188IbdHMw6AoAzzjgD1113HWbOnMm7nqFQCKqqwufzFX02PMHhh4gQjUa5pdzQ0IDly5fjo48+wpw5cxCNRkFEUFUVZWVlkGUZLS0tSCaTKC8vh2EYSCQSPOWrtWdovVYHDhyIK6+8EtOmTcMZZ5xhi/Hfs2cP1q9fj0suuQR+v5/fF/F23DSd4WjQn0I/KJz+HbSAx8v/rior53/73R7a+MWXRPvH9Zqbm+mpp56i0047jW9bbsize of out = 990, bytes = 990
D/BluetoothChatService: out = W0t/9vr9VJlZaUtORNgLxFlfe+yyy6jf/7zn5RKpfhU8EIhVMU+9Vtw+EmlUnml2oiIGhoaaOPGjfTLX/6S56VhLXcAHABPPsUqFKmqygcsrTlN3G43TZs2jX71q1/R1q1beTji1q1bbVXiu3v/Hg36IwYdu4giK9mQJcqgvKwczS3NcGpOBINB/P6Pf8SJJ56Iuc8+g7lz52Lnzp3w+XyQZRnhcBjpdBolJSXQdR26ruf5q6uqqrBt2zY+0ON0OjFp0iT84Ac/wJlnngnDMHgYYDwe58ViWRc2k8kUTFZvRVjYRzeRSMQ2QSoWi8EwDPh8Ptv4RWtrK/7+97/jd7/7HVauXAkgO9ho9ZHnFmQuKytDKBSyTeCyXuOSJGH48OG48sorMX36dKTTaVxzzTU8R7umaUjnrLMrHA36k4sQ7E7gkBzIUIa7R1zObLmsS6dMwbJly9DSGoJhGDwGlfmqXS4XHzkHwKuSG4bB/dRDhw4FAFxwwQWYNWsWnxoci8Xg9XrR0NDAY14Z1oEiEi4RQSdgA92FqtMzY4LF9a9YsQJ//OMfMX/+fMTjcf5dj8cDVVWRSCTyxJut2+12w+FwIJlM2qbFn3baaRgwYADeeOMNPoU+Ho/DKDCu0+l92r880vXHihDsTuD1eLM199JZHx6LGFE1Z1aQU/kXr6IocLvdiEQivLqLaZr8Qg8EAhg0aBCmTp2Km266CSeffDKA7MAQG7Rhos3+zmQy8Hq9tptOCLbgYND+qjWszqb19ba2NrjdbltPra2tDaFQCPF4HE8++SQ++OADfPLJJ1zUFUVBKpXik3I8Hg80TeMD5AxN0+Dz+dDS0gLgQLUm4IDvWww6dg0h2AdBUzUeS0r7ZxgahgECIQPA5XRBdWZr8bGLGMgmeSIiaJrG/wayQj1hwgR861vfwrRp0/jvxONxOJ1OyLIMwzCgqiokSSoYhxuLxfi25JbpykUI9tENK4jMrgN2jbJsize of out = 990, bytes = 990
D/BluetoothChatService: out = ID1Z8OTeOGzgQMrhu3To8/fTTeP3117FhwwZ+jVdUVCCZTNpqTHo8Hni9XrS1tUHXdS7MqqpyUQfAC0GLsL6uIckAZQA4kD0AXV325cECACZHh7r9B1t6XW7Ek0lIIEgOGR6XC5F4DE5FRZoIqbQJp9PJw/oAcLG1dhsrKytx9dVX46abbsKZZ54JWZaRTCYRCoVQXV3N41bT6TR3pbBJN6lUCoZhcJeKFWFhCzpDrnAD2bhta1Y9K6ZpIpPJ8Gno/7+9846Pq7ry+G/qm6YpGkmW5SJLGBdKAFMMAUxwp3mN4ZNlQwlLWQcSPvFSHHbzARbYZMluYBeTAKGD6cXJx3bAy+IPMdhggzFuWG5yk211aTSj6Zo5+8dwjt6TZSRLBnnQ/X4+7/NUZt679777zj333HPOZUVi0aJFePrpp7F69WoJzNL3d+6PJpNJMgfGYjEx6emzVabT6X6/v993+dP1bDIBRF/fuC/nY4W+lr+/Z32uj+6YOHEiLr30Ulx55ZUYN24cAEjaS55OKhTHKul0WoQta9ytra346KOP8Le//Q0LFiyQ/u92u2EymQwad3c2cz0SRAYlf/Kx3nmJ2+2G3+9Hc3MzEokECgoKMGvWLMyYMQOTJk1CWVnZYSMKu8tXrFB8V3yTsgFAzHr0dWQsm+ASiYR4RT333HN47LHHUF9fDyCXVCqVSiEWi/VaYCt6hxLY/cTv9yMUCgHI2acvueQSzJ49GxMnTsSQIUNgs9kkmxkHH+hDiTkKTKE41kkmkzCbzQblo76+HoWFhWhubsbbb7+NJ598El999RUAIBgMorW19RuvqQT2kaEE9lFgzJgxmD17NubMmYOJEyca/sf2u8Np2KrDKgaSntZAUqkUnE4ngFxf5oVEXkzksHf95z///HP893//N9555x2lYR9lul9xUBwRXq8XXq8XNTU1qK6uRnNzsywecn4QPujrLZQcDgc0TevxhVEovk16Epjpdsize of out = 990, bytes = 990
D/BluetoothChatService: out = FpC2jkwhj2iUqmUeJzwdWw2G0KhEEwmE0pKSiTXtuLooDTsfqLfYonRpzOlrzcO4ATv7EnCn+uacEehOJbhnDlEJNuPpdNp6cccPMPBYUrDProogX2U8Hq9kl5SHx3GiXI0TYPL5ZJEPGzL7smPWqH4NumNwOR87Kx8MHoPKa/XC4vFgnA4LJ/x+/2GxGd9vb+iEyWw+4k+PSrQuY0Xa9Rds/Xps5txUI1CMVD01iTncOTSMSSTSZklct715uZmALm+z7NGVlJ66t9KYB8ZSmAr+kWv/Wy/nkrzhgwsKL7tF7a/1++pft/2/fXoN7PommjpWEIfnNPT7kZ6E6GiZ9Sio+I7Qy9k9BFxiiPjWBbWim8XJbAV3yp6gdydtvltC+x8HxC6Jvo61oR1T+17LJX1+4AS2IpvlcO9sPrp/bfJ90Fgf9PMZKAFIg8oXcvHPw90+b5v9M9Ap1D0AbPZDKvVqjYA7gVd7db8N7PZ3KN9+LuA1yX06xPAwA8k31eUhq34VumqIQKdQqi/C3q9ob8a9kALnu7MICyszWbzgC/adV1E7tpeA91+3zeUwFZ8qzidTkk9y5Ge2Wz2OzNV5LvA1vtAA53aNW9G0N2u5t8l+l2POAqyq1lEcfQwc3fu61mh+CY4BJ+3NAM6oz8HWjvMB7qzA7PQ/i5mKL2BB8WBXC8YNPKLdx/u6xl5fpSXlxMAuuCCC0jTNPL7/TRixAgaOXIkuVyuQ3Y5592ibTabYafowx0Wi4XMZjOZzWYymUyH/J//19ejp/v39H2TyURms5ksFgtZrVayWq1ksVikrD193263U2Fhodxv5MiRBICeeeYZqq+vp6amJlq3bh1dffXVBIC8Xi+VlpbKz719Tly2rm3bl2eur3PX3/Vt0Jv21TSNgsEglZaWUkVFBQEgt9tNDofjkPsdrg9wXTweD5WUlNBxxx1HAMhut0vdAZDf7ycAdP3119P+/fuptbWVXnjhBSosize of out = 990, bytes = 990
D/BluetoothChatService: out = rKyMAVFxcTJqmyXd6U34uX3d1N5lMZLfbDe1eUlIi78SIESPotddeo0wmQ5s3b6ZJkyZJn7HZbIeta9d76/sqv1v6PviN38cgk1/5vs380TiCwSDt27eP6urqKB6PUywWo3g8Ts3NzeTz+Qwd32azScfml6M/x0ALbIvFIkd31+vN/cvKyuQlttls9Nvf/pZqa2uJiKilpYWIiBoaGuiiiy6Sa7Iw+raPw730eiGqb4tvEqrdHZMmTaK6ujratGkT7du3j+bPny//YwGrv19313Y4HHTJJZfQli1bKJ1O07p160Ro22w2KisrI4vFQj6fj9xuN1VVVRna9tprryWbzUZut1uuqWlarwV21+et7xNATvjzAMRKisvlon/9138lJplMUigUIiA3EHu93l73zyNp767HYJNfx8acaoAxm80oKyvDkCFD4HA4kEwmDTudd/WFzWazkn1voOFFn8MdPaFf1OqrvbG2thbJZBJutxtWqxW33HKL7KTj9XqRTqdRXFyMK664AkDOLtvbnXb0ppS+wKaD7tqCba4MpxM4knZIp9MYMmQITjrpJIwYMQI/+clPMGrUKAAwpCTQty+XibM2JhIJVFdXY+TIkbBaraisrBQ7dTqdRkNDAzKZDNrb25HNZmXnIs6lXl5eDk3TDEnI9Bn0joTu+kNHRwfsdjtcLpeEmrvdbowcORJEhKamJrS0tMDn80m9w+Fwr+7PqVqPFfPOsc6gb6XCwkIUFhYimUwiHo8DyCWt8fl8KCwslA6lp7tEOIfDbrfDbrfLRgZHu2P2V2CzkOKXy2KxwGq1wm639yoxVVFRkezKTURwuVzw+/2SZpOI0NraKjuXBINBAL3facdisRjqwQn0u+5PeDi+SWCzUOuLoGZ8Ph/q6+sRj8cRCoUwevRo/O53v4PP55MydufqxnZoXjSMxWIyiLW1taGhocGQdxqAtOvrr7+OzZs3AwA+/PBDrFu3Du3t7bIxssize of out = 990, bytes = 990
[... cut ...] lwJSTTz5ZbJZdy7Np0yYaP368od2nTZtGCxYsoA0bNlA0GqVkMkmrVq0yaH8FBQV08sknS7lDoZDMhurr6+m0004zCCkA9E//9E9SDrZJExHdf//90s488HF2QrPZTLNmzSKiTs19yJAhBEBmH7feeisdOHCAiHKmpDvuuEOEMWuU3ySwuwbDAKDbbrtNBv9t27YR0GmysNvt0hbBYFDqwcK3a725zfSLyPfff79cf+PGjWJq4ufM/aOgoMCwKD1z5kzSNI18Pp+YTYCc+Ydng3wPq9VKo0aNMtilNU0Trbu0tJR27NghsxdeN9Gv35jNZiWwv/cV7nL0RsPuKpCAXIY6fsH1K9wNDQ3iudDVo+DCCy+UKaReCBKRCAGTyST3+/jjj+W6ehPLTTfdJB2bX55NmzaJs

```

A lot of `E/NewNetworkFragment: Result not Found` errors are being spitted out.

sydneyaldo, 2020-06-10:
I have the same error as well

rrijal53, 2021-05-22:
duplicate https://github.com/treehouses/remote/issues/800

closed 2021-05-22 by rrijal53