treehouses/remote #851: Changing colors for Settings Fragment

ali18997, 2020-06-11:
Changing background and text colors through **app_preferences.xml** doest work as it does for other layouts. One way to change background is through **SettingsFragment.kt** but changing text color is still challenging through that approach. Is there a better way to go about this?

sriharivishnu, 2020-06-18:
- Best direction to go for this is to make Custom Preference View / Preference Categories and override the onCreate or onBind method and then set the text colour from there (depending on in night mode or not). 

- Settings screen Fragment background can be set by setting the root view background colour.

sriharivishnu, 2020-06-18:
[Reference](https://stackoverflow.com/questions/11607302/how-to-change-text-color-of-preference-category-in-android)

PR #996 (2020-07-01): settings: dark mode (fixes #851)
# Fixes #851 

[image]
commit 85ea3ce5 (2020-07-01): settings: dark mode (fixes #851) (#996)
closed 2020-07-01 by dogi