treehouses/remote #880: new release crashes when Terminal is visited

law-lin, 2020-06-18:
Every time I connect and open the Terminal tab, the app crashes.

I'm using a OnePlus 6T.

law-lin, 2020-06-18:
Reinstalled it three times, tried new image release (135), Terminal crashes app every time.

sriharivishnu, 2020-06-18:
```
java.lang.IllegalArgumentException: Parameter specified as non-null is null: method kotlin.jvm.internal.Intrinsics.checkParameterIsNotNull, parameter convertView
        at io.treehouses.remote.adapter.CommandListAdapter.getGroupView(Unknown Source:2)
        at android.widget.ExpandableListConnector.getView(ExpandableListConnector.java:446)
        at android.widget.AbsListView.obtainView(AbsListView.java:2405)
        at android.widget.ListView.measureHeightOfChildren(ListView.java:1421)
        at android.widget.ListView.onMeasure(ListView.java:1327)
        at android.view.View.measure(View.java:24545)
        at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6828)
        at android.widget.LinearLayout.measureChildBeforeLayout(LinearLayout.java:1552)
        at android.widget.LinearLayout.measureVertical(LinearLayout.java:842)
        at android.widget.LinearLayout.onMeasure(LinearLayout.java:721)
        at android.view.View.measure(View.java:24545)
        at androidx.constraintlayout.widget.ConstraintLayout$Measurer.measure(ConstraintLayout.java:736)
        at androidx.constraintlayout.solver.widgets.analyzer.BasicMeasure.measure(BasicMeasure.java:399)
        at androidx.constraintlayout.solver.widgets.analyzer.BasicMeasure.measureChildren(BasicMeasure.java:105)
        at androidx.constraintlayout.solver.widgets.analyzer.BasicMeasure.solverMeasure(BasicMeasure.java:232)
        at androidx.constraintlayout.solver.widgets.ConstraintWidgetContainer.measure(ConstraintWidgetContainer.java:113)
        at androidx.constraintlayout.widget.ConstraintLayout.resolveSystem(ConstraintLayout.java:1480)
        at androidx.constraintlayout.widget.ConstraintLayout.onMeasure(ConstraintLayout.java:1555)
        at android.view.View.measure(View.java:24545)
        at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6828)
        at android.widget.FrameLayout.onMeasure(FrameLayout.java:194)
        at android.view.View.measure(View.java:24545)
        at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6828)
        at androidx.coordinatorlayout.widget.CoordinatorLayout.onMeasureChild(CoordinatorLayout.java:760)
        at com.google.android.material.appbar.HeaderScrollingViewBehavior.onMeasureChild(HeaderScrollingViewBehavior.java:99)
        at com.google.android.material.appbar.AppBarLayout$ScrollingViewBehavior.onMeasureChild(AppBarLayout.java:1892)
        at androidx.coordinatorlayout.widget.CoordinatorLayout.onMeasure(CoordinatorLayout.java:831)
...
```

PR #882 (2020-06-18): terminal: no crash (fixes #880)
# Fixes #880 

## Issue
- During kotlinification, convertview was made from a nullable type to a non-nullable type

## Fix
- convertView can be null, so make it a nullable type
commit eb21982f (2020-06-18): terminal: no crash (fixes #880) (#882)
closed 2020-06-18 by dogi