treehouses/remote #900: Services wants user to update even when updated

dyeh123, 2020-06-19:
Services fragment wants user to update CLI even when updated.
[image]
[image]

sriharivishnu, 2020-06-20:
The dialog pops up when there is an error: if you could provide Log of the error, could help in debugging. 

We could have a dialog for error, but then to detect whether the CLI version is outdated, we would have to run `treehouses remote version`.

dyeh123, 2020-06-20:
Here is the [log](https://gist.github.com/dyeh123/cec220e5128d281df5f53dd266aa4bca)
Possible culprits are line 535, line 7 or line 123?

PR #940 (2020-06-24): services: update cli handling (fixes #900)
# Fixes #900 

- Alert dialog does not show anymore (improved the error checking)
commit c5a2cc08 (2020-06-24): services: update cli handling (fixes #900) (#940)
closed 2020-06-24 by dogi