treehouses/remote #913: Closing Phone Bluetooth when Pi is connected causes app to crash

ali18997, 2020-06-20:
[image]
[image]
[image]

sydneyaldo, 2020-06-20:
I thought I fixed this, is this on master?

ali18997, 2020-06-20:
> I thought I fixed this, is this on master?

I was testing this on master, I tried it on your fix and somehow that fix solves this as well, even though it felt like a different issue

sydneyaldo, 2020-06-24:
Tried on master and it didn't crash

sydneyaldo, 2020-06-26:
Error happens on certain fragments such as `System`