treehouses/remote #1476: security :  use shared preference in private mode

rrijal53, 2020-08-27:
Use shared preferences in private mode. 
```
val sharedPref = activity?.getPreferences(Context.MODE_PRIVATE)

```

sriharivishnu, 2020-08-29:
@rrijal53 The default value is MODE_PRIVATE. 
```
/**
     * Gets a {@link SharedPreferences} instance that points to the default file that is used by
     * the preference framework in the given context.
     *
     * @param context The context of the preferences whose values are wanted
     * @return A {@link SharedPreferences} instance that can be used to retrieve and listen to
     * values of the preferences
     */
    public static SharedPreferences getDefaultSharedPreferences(Context context) {
        return context.getSharedPreferences(getDefaultSharedPreferencesName(context),
                getDefaultSharedPreferencesMode());
    }

    private static String getDefaultSharedPreferencesName(Context context) {
        return context.getPackageName() + "_preferences";
    }

    private static int getDefaultSharedPreferencesMode() {
        return Context.MODE_PRIVATE;
    }
```

rrijal53, 2020-08-29:
> @rrijal53 The default value is MODE_PRIVATE.
>
> ```
> /**
>      * Gets a {@link SharedPreferences} instance that points to the default file that is used by
>      * the preference framework in the given [...]

oh ok , i thought it will use other mode. Closing the issue.

closed 2020-08-29 by rrijal53