treehouses/remote #1488: Incorrect title on Community Map

anthonyhvo12, 2020-08-29:
**Problem**
The title of the view remains as "Services" after switching from Services to Community view.

**Screenshot**
[image]

dyeh123, 2020-08-31:
Could you provide steps on how to reproduce that? Switching from services to community replaces the title for me.

BryanGazali, 2021-01-25:
I have found a way to replicate this issue. If you go from Services -> Community -> Settings, and then use your phone's native back button to go back to Community, it should show Service in the title. I also found that this issue similar happens with other pages. For example, here I went Home -> Terminal -> Community, and then returned to Terminal with my phone's back button. The Terminal title now shows Home
[image]