treehouses/remote #1935: How to connect to Raspbian Bookworm?

paolosezart, 2024-02-22:
I have a Raspberry pi4B with Raspbian Bookworm operating system installed, I would like to connect my Xiaomi Redmi note 8pro to it via bluetooth. The treehouses program detects RPI via bluetooth, but it won't connect:

[image]
[image]

What else do I need to configure on Raspbian that I can control RPI via bluetooth?

Okuro3499, 2024-03-08:
@paolosezart what is your device android version?

paolosezart, 2024-04-05:
I have android 9

dogi, 2024-05-14:
@paolosezart the newest version of remote is now able to connect to our new bookworm image in treehouses/cube